analogdevicesinc/scopy #1344: Cannot install Scopy Android apk on Samsung Galaxy Tab A (SM-T510)

pborring, 2022-05-21:
**Environment:**
Samsung Galaxy Tab A (SM-T510) with Android 11, and Samsung OneUI 3.1

**Describe the bug**
I am trying to install the new [Scopy v1.4.0](https://github.com/analogdevicesinc/scopy/releases/tag/v1.4.0) release on my Samsung Galaxy Tab A, but it fails despite enabling to install from source.

**To Reproduce**
I have tried two different paths

Direct install on tablet:
-------------------------
Download apk in browser
Invoke install, tried browser download (used Edge) as well as Samsung My Files file explorer
Tablet asks for permission to trust source > granted
Install starts
Install terminates with "App was not installed"

As no error message/log in above, I have enabled development mode/debug on tablet and installed Android Debug Bridge on Windows PC and connected through USB to tablel.
* Calling adb -d install ..\Scopy-v1.4.0-android.apk
* Results in:
Performing Streamed Install
adb: failed to install ..\Scopy-v1.4.0-android.apk: Failure [INSTALL_FAILED_NO_MATCHING_ABIS: Failed to extract native libraries, res=-113]

**Expected behavior**
Well, it would be nice to be able to install on the Samsung Galaxy Tab, I assume it is one of the more common Android tablets

**Additional context**
* Searching the net it seems is lack of inclusion of target platform in build 
* Disabling Google Play scanning does not make a difference.
* Installing APK on my Samsung Galaxy S10e running Android 12 with Samsung One UI 4.1 works, although app does not render correctly on phone screen. I assume phone is not a target UI anyway though, if it is I can provide more context/screenshots in a different bug.

adisuciu, 2022-05-23:
Hi, 
I think your system is running a 32bit OS/CPU. Can you run 

cat /system/build.prop in adb shell 
or
adb shell getprop ro.product.cpu.abilist

to check the CPU ABI and post the result ? 

I'm trying to get a 32bit apk meanwhile ..

adisuciu, 2022-05-23:
Can you try this ? 
https://swdownloads.analog.com/cse/scopy-v1.4.0-android-arm32.apk

pborring, 2022-05-24:
That private build now installs correctly! The app is a bit slow/sluggish compared to Windows laptop, but that might just be the tablet being underpowered. Let me know if you'd like any perf details, but I will close this entry as fixed, given that it now installs. Thanks.

adisuciu, 2022-05-24:
That's alright. I'll upload it to the release page. Thanks for testing !

closed 2022-05-24 by adisuciu