analogdevicesinc/scopy #1346: ADALM2000: <crashes when using the osiclloscope>

donharter, 2022-06-05:
OS Version: Linux 5.7.2-050702-generic
FW Version: 
SW Version: 6f32056

Description of the bug:<description>
Steps to reproduce:
set up an output signal
connect analog input 1 to the output signal
turn on triangle wave
try to view it in the oscilloscope
-
 flatpak run org.adi.Scopy

Note that the directories 

'/var/lib/flatpak/exports/share'
'/home/drh/.local/share/flatpak/exports/share'

are not in the search path set by the XDG_DATA_DIRS environment variable, so
applications installed by Flatpak may not appear on your desktop until the
session is restarted.

GLib-GIO-Message: Using the 'memory' GSettings backend.  Your settings will not be saved or shared with other applications.
ERROR: Unable to claim interface 2:4:5: -16

(scopy:2): GLib-CRITICAL **: g_variant_equal: assertion 'one != NULL && two != NULL' failed
QFutureWatcher::connect: connecting after calling setFuture() is likely to produce race
QMetaObject::connectSlotsByName: No matching signal for on_btnAuto_toggled(bool)
QMetaObject::connectSlotsByName: No matching signal for on_xyPlotLineType_toggled(bool)
QObject::connect: invalid null parameter
QMetaObject::connectSlotsByName: No matching signal for on_nextButton_clicked()
QMetaObject::connectSlotsByName: No matching signal for on_restartButton_clicked()
QMetaObject::connectSlotsByName: No matching signal for on_finishButton_clicked()
terminate called after throwing an instance of 'std::runtime_error'
  what():  moving_average_ff(279): insufficient connected input ports (1 needed, 0 connected)

I see no .ini file anywhere.

The ini files might be useful to reproduce the error.
The ini file is located at: [/home/drh/.var/app/org.adi.Scopy/config/ADI/Scopy.ini](url)
Please consider attaching it.

adisuciu, 2022-06-06:
Does this reproduce on every run ? Can you share the ini file ?

adisuciu, 2022-06-06:
Also it looks like you are using an older version of Scopy .. Consider updating to the latest - https://github.com/analogdevicesinc/scopy/releases/tag/v1.4.0

donharter, 2022-06-06:
not the latest version

donharter, 2022-06-06:
see the new bug report with the latest version

closed 2022-06-06 by donharter