analogdevicesinc/scopy #1352: ADALM2000: errors read line, read integer

donharter, 2022-06-22:
**Environment:**
 - OS: ubuntu 20.04.4 LT
 - FW 0.28
 - SW Version 1.4.0
 - 
[Preferences_ini.txt](https://github.com/analogdevicesinc/scopy/files/8960944/Preferences_ini.txt)
[Scopy_ini.txt](https://github.com/analogdevicesinc/scopy/files/8960945/Scopy_ini.txt)

The below errors repeat.
~$ flatpak run org.adi.Scopy

Note that the directories 

'/var/lib/flatpak/exports/share'
'/home/drh/.local/share/flatpak/exports/share'

are not in the search path set by the XDG_DATA_DIRS environment variable, so
applications installed by Flatpak may not appear on your desktop until the
session is restarted.

QIODevice::read (QFile, ":/stylesheets/themes/global.qss"): device not open
QIODevice::read (QFile, ":stylesheets/stylesheets/browserStylesheet.qss"): device not open
Warning: Using function expressions as statements in scripts is not compliant with the ECMAScript specification:
 "function(o) { for (each in o) { print(each); } }..." 
This will throw a syntax error in Qt 5.12. If you want a function expression, surround it by parentheses.
Unknown property link-color
Unknown property link-color
ERROR: Unable to claim interface 2:6:5: Device or resource busy (16)
QMetaObject::connectSlotsByName: No matching signal for on_xyLineThickness_currentIndexChanged(int)
QMetaObject::connectSlotsByName: No matching signal for on_xyPlotLineType_toggled(bool)
QMetaObject::connectSlotsByName: No matching signal for on_nextButton_clicked()
QMetaObject::connectSlotsByName: No matching signal for on_restartButton_clicked()
QMetaObject::connectSlotsByName: No matching signal for on_finishButton_clicked()
QObject::connect: No such slot adiscope::SpectrumAnalyzer::onChannelAdded(int)
QObject::connect:  (receiver name: 'SpectrumAnalyzer')
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 40000. Due to alignment requirements
   64 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 40000. Due to alignment requirements
   64 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
ERROR: READ LINE: -9
ERROR: READ INTEGER: -9
E20220622 09:44:08.124722   175 buffer.cpp:481] Buffer: Cannot refill RX buffer
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 40000. Due to alignment requirements
   64 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 40000. Due to alignment requirements
   64 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
ERROR: READ LINE: -9
ERROR: READ INTEGER: -9
E20220622 09:44:08.182243   229 buffer.cpp:481] Buffer: Cannot refill RX buffer
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 40000. Due to alignment requirements
   64 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 40000. Due to alignment requirements
   64 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
ERROR: READ LINE: -9
ERROR: READ INTEGER: -9
E20220622 09:44:08.241993   284 buffer.cpp:481] Buffer: Cannot refill RX buffer
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 40000. Due to alignment requirements
   64 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 40000. Due to alignment requirements
   64 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
ERROR: READ LINE: -9
ERROR: READ INTEGER: -9
E20220622 09:44:08.450275   338 buffer.cpp:481] Buffer: Cannot refill RX buffer
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 40000. Due to alignment requirements
   64 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 40000. Due to alignment requirements
   64 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
ERROR: READ LINE: -9
ERROR: READ INTEGER: -9
E20220622 09:44:08.496510   392 buffer.cpp:481] Buffer: Cannot refill RX buffer
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 40000. Due to alignment requirements
   64 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 40000. Due to alignment requirements
   64 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
ERROR: READ LINE: -9
ERROR: READ INTEGER: -9
E20220622 09:44:08.540275   446 buffer.cpp:481] Buffer: Cannot refill RX buffer
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 40000. Due to alignment requirements
   64 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 40000. Due to alignment requirements
   64 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
ERROR: READ LINE: -9
ERROR: READ INTEGER: -9
E20220622 09:44:12.339866   500 buffer.cpp:481] Buffer: Cannot refill RX buffer
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 40000. Due to alignment requirements
   64 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 40000. Due to alignment requirements
   64 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
ERROR: WRITE ALL: -9
E20220622 09:44:12.385839   554 buffer.cpp:481] Buffer: Cannot refill RX buffer
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 40000. Due to alignment requirements
   64 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 40000. Due to alignment requirements
   64 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
ERROR: READ LINE: -9
ERROR: READ INTEGER: -9
E20220622 09:44:12.440263   608 buffer.cpp:481] Buffer: Cannot refill RX buffer
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 40000. Due to alignment requirements
   64 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 40000.

donharter, 2022-06-22:
I am also getting noise on channel 2.  It should be a triangular wave.  I just have 3 resistors in series. the first one is a current limiter. the second one is the load. and the third one is a current sensing resistor, 330 ohms,load resistor and 1 ohm resistor.
Source voltage is a triangular wave on channel 1. Channel 2 should look similar but has noise which messes up the xy plot screen.

adisuciu, 2022-06-23:
The readline/ read integer errors do not affect measurements in any way. Can you share a screenshot of the noise ?

donharter, 2022-06-23:
zoomed in on the waveform
[image]

adisuciu, 2022-06-23:
The voltage on channel 2 peak to peak is less than 10mv. The 12bit ADC gives a resolution of 1.4mV LSB . So in your screenshot you have about 4 bits worth of data. A few bits of noise will always be present on top of that data.

If this is indeed what you are trying to measure, the adalm2000 is not the tool to measure that. If not I would recheck the connections / computations.

No bug here in my opinion.
-Adrian

adisuciu, 2022-06-23:
You can also check out this post - https://ez.analog.com/studentzone/f/q-a/118735/adalm1000-adalm2000-current-and-voltage-measure-resolution

donharter, 2022-06-23:
I increased the input waveform to 10V and it made a difference on the oscilloscope display.  I can't use that since it would possibly fry some digital chips. 
[image]

donharter, 2022-06-23:
I would have expected the adalm2000 to have some builtin amplifiers.  Next model they should add that.