analogdevicesinc/scopy #1372: <Instrument>: <Title>

jpnielsen, 2022-09-19:
Please support Google Android TV / Nokia Streaming Box 8000

I would like to run scopy on this inexpensive Android box - but it's says it's not supported when installing from the play store.

https://www.nokia.com/shop/streaming-devices/nokia-streaming-box-8000/

It says it's running "Android 10" - but it must be some special Google TV version.

Could you include it as s built-target - it should be fast enough I think.

adisuciu, 2022-09-19:
It's not really meant to run on android-tv. According to the documentaiton this is a bare minimum that's needed for support . https://github.com/analogdevicesinc/scopy/tree/android-tv - If this builds correctly you can give it a try.

adisuciu, 2022-09-19:
It's not really meant to run on android-tv. According to the documentaiton - https://developer.android.com/training/tv/start/start - this is a bare minimum that's needed for support . https://github.com/analogdevicesinc/scopy/tree/android-tv - If this builds correctly you can give it a try.

adisuciu, 2022-09-19:
According to this, you should already be able to sideload it - https://forum.qt.io/topic/70512/qt-application-for-smart-tv/4 
The modification from android-tv branch will just list it in the menu/google store .. I don't have such a device so I can't test. If you want, you can side load it and report back.

closed 2024-09-24 by adisuciu