analogdevicesinc/scopy #1376: Scopy: missing top menu bar

acusco, 2022-10-19:
**Environment:**
 - OS: Linux (Ubuntu 22.04)
 - FW Version: v0.28
 - SW Version or git tag:  1.4.1-f4beeb1

**Describe the bug**
When running scopy, the top menu bar including the minimize, restore and close buttons are missing. 
Also, dragging or moving the window around not working.

**Screenshots**
[image]

Modiug, 2022-10-23:
Same here.

I wonder if it might be related to the use of Wayland. [Something similar was reported for MPV](https://ubuntuhandbook.org/index.php/2022/06/mpv-no-title-bar-border-ubuntu-2204/).

Modiug, 2022-10-24:
Indeed, switching Ubuntu to one of the "on Xorg" desktops (and not using Wayland) will bring back the window decoration. However, that's only a workaround.

adisuciu, 2022-10-24:
I'm not sure if this is our bug or Wayland bug or both. We tried fixing this previously, but it seems it didn't stick ...

Regardless, workaround is running Scopy like this: 

XDG_SESSION_TYPE=x11 flatpak run org.adi.Scopy

We will take another look before a new release.
-Adrian