analogdevicesinc/scopy #1377: Scopy: Logic Analyzer no longer shows decoded data at changed baud rate

SgoSkz, 2022-10-29:
**Environment:**
 - OS: Windows 10
 - FW Version: v0.25
 - SW Version: v1.4.1-f4beeb1
 
**Describe the bug**
Logic Analyzer no longer shows the decoded data for UART when changing baud rate

**To Reproduce**
Steps to reproduce the behavior:
1. Go to "Logic Analyzer" in Scopy
2. Set UART to respective pins
3. Set single capture
4. See decoded data
5. Change baud rate
6. See error

**Expected behavior**
Decoded data shows up

**Screenshots**
[image]
115200 (default) baud rate
[image]
38400 (changed) baud rate

**Additional context**
Scopy.ini
https://pastehub.net/fdc83de4341

SgoSkz, 2022-10-29:
Updating firmware did not fix the issue.
New FW Version: v0.29

adisuciu, 2022-11-03:
I don't think it should decode though. To me It looks like your signal is at 115200 baudrate, not at 38400 ..

SgoSkz, 2022-11-03:
You are absolutely correct, 115200 is the correct baudrate, but when I change it back to 115200 from 38400, the decoded messages still dont show up. I'll add a screenshot soon

SgoSkz, 2022-11-04:
[image]

jjmoreno1163, 2022-11-04:
Hi, I have the same issue, when changing to any baudrate value, decoding doesn't work.

adisuciu, 2022-11-16:
The solution here would be to trigger another decoding when one of the decoding parameters changes if I understand correctly. It should do this even when the acquisition stopped.

SgoSkz, 2022-11-16:
Could you elaborate a little more on that? From what I'm getting, you mean to scan for another message and it should decode properly? That is not the case for my situation, I will provide screenshots when I get the time.

adisuciu, 2022-11-21:
Well it seems to me that there are no new acquisitions in betwseen changing baud rates.
Decoding only seems to work upon new acquisition, not upon changing decoding parameters. My suggestion is to either decode upon changing decoding parameters, or manually trigerring a new decode (via a button).

SgoSkz, 2022-11-21:
This is an active 9600 baud rate scan with no decode messages at 9600 decode value for the logic analyzer.
[image]
Changing any respective trigger settings for both DIO and extern trigger does not cause decoded messages to appear

adisuciu, 2022-11-22:
I just tested and it works just fine for me .. 
[image]
Can you try removing your ini files ? https://wiki.analog.com/university/tools/m2k/scopy-troubleshooting?s[]=scopy&s[]=troubleshooting#reinstall_ini_files

SgoSkz, 2022-11-22:
unfortunately, deleting the ini files in %AppData%/ADI does not solve the issue for me. Is the version you are testing also v1.4.1?

[image]
[image]

adisuciu, 2022-11-22:
I reproduced this. I used v1.4.0 where things seem to work just fine. I will take a look.

pitsche, 2023-10-12:
I found a workaround.
Chanche the default in C:\Program Files\Analog Devices\Scopy\decoders\uart\pd.py to the value you need.
Restart Scopy and that's it.

molitoris, 2024-06-25:
The logic analyzer does no longer shows the decoded data for UART after changing the baud rate

OS: Windows 11
SW Version: v1.4.1-f4beeb1

andrei47w, 2024-06-25:
This bug was fixed for all platforms on the main branch. Will be available in the next release.