analogdevicesinc/scopy #1381: Scopy: Implement "Roll" Mode

muhuna, 2022-12-30:
This is regarding the oscilloscope part of Scopy. When working with low frequency signals and not knowing exactly what issue one is dealing with it can often be very helpful to see a "live" stream of the samples captured. Actually, with other PC based scopes I had the chance to work with I sometimes just use a manual trigger by hitting the "stop recording" button or manually change modes of the HW under test when something happens. Not always but sometimes it could be done with trigger and some program to change modes on the HW if there is an API. But that just adds a lot of time for doing something pretty simple. Unfortunately, this isn't currently possibly with Scopy because there is no "live" updated (in my case I sometimes have to wait up to 8s to get an updated screen). Hence, it would be nice to have a "roll" mode.

rgetz, 2023-01-03:
"roll" mode is plotting available data, as it comes in, without discarding the "old" data that shifts left on the screen. The "roll" rate (the rate that the shift happens) depends on the capture speed. (the faster you capture, the more you can shift over; the slower you capture, the less data you capture, the slower the shift).

For a capture rate of 100 Hz, you may only want to capture 10 samples (shift the previous data over those 10 samples), and have a screen update rate of 10 Hz... The slower the sample capture, the less samples you want to capture, to keep the screen update to a smooth rate. (you may get to a point where you only want one sample between updates).

[image]

rgetz, 2023-01-04:
yeah, blanking the screen at the end (rather than shifting it over) is one solution.

I would suggest to play with a real scope in the lab - to see how things are managed as sample rate decreases, and then try to emulate that. :)

rgetz, 2023-01-09:
@damercer 

yeah, older versions of waveforms did this...

The min/max was plotted behind as a shaded version of the colour, but the average was plotted as the line.

[image]

The described it as : "Noise Band: shows or hides the noise (min/max values)." which I don't think actually captures things properly.

[image]

The above frequency sweep might look like damped signal (C1), but noise band on R1 indicates that this is not true. 
This indicates glitches or high frequency components that might be hidden by the averaging...

what it does - and how it's described - is super important.

rgetz, 2023-02-14:
A good example is https://forum.qt.io/topic/84532/qml-animation-trail-on-object