analogdevicesinc/scopy #1384: ADALM2000: No decoder menu

crides, 2023-01-13:
**Environment:**
 - OS: Arch Linux
 - FW Version: v0.25
 - SW Version: v1.4.1, installed from flatpak from release

**Describe the bug**

Scopy doesn't show decoder menu

**To Reproduce**
Steps to reproduce the behavior:
1. connect to adalm2000
2. go to Logic Analyzer tab

**Expected behavior**

The decoder menu is on the right side of the trigger menu [like in here](https://wiki.analog.com/university/tools/m2k/scopy/logicanalyzer)

**Screenshots**

[image]

adisuciu, 2023-01-13:
Did you double click the newly added decoder (I2C) ? It should be highlighted as here: https://wiki.analog.com/_detail/university/tools/m2k/scopy/logic_analyzer_new_6.png?id=university%3Atools%3Am2k%3Ascopy%3Alogicanalyzer 
(notice the white highlighting around the decoder channel handle) ?

If you mean the decoder menu as here - https://wiki.analog.com/_detail/university/tools/m2k/scopy/logic_analyzer_new_14.png?id=university%3Atools%3Am2k%3Ascopy%3Alogicanalyzer 
That will be available only in the next version of Scopy - the documentation got ahead of the releases - sorry about that :) 

-Adrian

crides, 2023-01-13:
I do mean the second menu. The release notes does mention supporting exporting decoded signals, how should that be done then?

adisuciu, 2023-01-13:
https://wiki.analog.com/university/tools/m2k/scopy/logicanalyzer?s[]=logic&s[]=analyzer#export
Regular export button in the general settings, should now have "decoder" channels along with the binary channels. You can export them from there. 
There's a preference option in the Logic Analyzer "Separate decoder annotation when exporting" - this controls the format in which the decoder annotation are exported - one annotation per sample or a a format that has a begin and an end "tag" for the annotation. You can experiment with that .. 
-Adrian

crides, 2023-01-20:
Attached is a video of me recording some i2c traffic. The signal is decoded succesfully, but I can't find any other channel in the export drop down, and there are no other channels if I export all channels either (to vcd, exporting to CSV hangs basically forever (>20h))

https://user-images.githubusercontent.com/15631555/213815013-1c2b2c1d-f771-408a-bb25-1c925e038bf4.mp4

adisuciu, 2023-02-07:
We reproduced that error, and it looks like we fixed that during decoder menu development. Can you try this development build ?
https://github.com/analogdevicesinc/scopy/suites/10306953295/artifacts/534918129 - this would be the direct download link
or here:
https://github.com/analogdevicesinc/scopy/actions/runs/3892616414