analogdevicesinc/scopy #1442: Scopy: Logic Analyzer SPI decoder not displayed for different SPI modes

kseerp, 2023-08-15:
**Environment:**
 - OS: Windows
 - FW Version: v0.29
 - SW Version or git tag: v1.4.1 - f4beeb1

**Describe the bug**
The logic analyzer SPI decoder will not appear for different SPI modes other than SPI mode 0 (CPol: 0, CPha: 0), even if the MCU SPI mode was set to its respective modes. Another scenario is that after switching the channel settings from other SPI modes (1, 2, 3) back to SPI Mode 0, the decoded data will no longer be displayed.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to "Logic Analyzer" in Scopy
2. Set Channel Settings to respective pins.
3. MCU SPI mode set to mode 3 (CPol: 1, CPha: 1).
4. Set Single Capture with CS triggered to falling edge.
5. See error

**Expected behavior**
Decoded data would be displayed for different SPI modes.

**Screenshots**
MCU SPI mode set to mode 3, scopy cpol and cpha set to 1.
[image]

Decoded data shows up when MCU SPI mode set to 0, scopy cpha and cpol set to 0.
[image]

Decoded data would no longer show after changing from different SPI modes back to SPI mode 0.
[image]

Brandon-Hurst, 2024-04-16:
Hello,

I have frequently run into this issue myself. Is there any way we can expedite getting this looked at?

Brandon-Hurst, 2024-06-04:
Documenting this is fixed in a development version: https://github.com/analogdevicesinc/scopy/actions/runs/8718825871/artifacts/1421340517

Brandon-Hurst, 2024-06-21:
Hello, 

Just pinging this issue again as it's been a while since this was put in place.

closed 2024-12-11 by AlexandraTrifan