analogdevicesinc/scopy #1452: Scopy: SPI generating bad patterns

lysynski, 2023-08-24:
**Environment:**
 - OS: Windows
 - FW 0.29 
 - SW Version v1.4.1

**Describe the bug**
SPI pattern generator creates bad messages when CPHA is set to 0. Bad data and missing final clock pulse.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Pattern Generator
2. create a group with 3 channels (bits); CLK, MOSI, CS for SPI pattern
3. set CPOL = 0, CPHA = 0, CS Polarity = 0, MSB First enabled, 2 Bytes per frame, data = 0500
4. See error (screenshot CAP1)
5. See analyzer capture - decoded data bad
6. set data=07FF
7. See error (screenshot CAP2)

**Expected behavior**
correct SPI pattern for setting applied

**Screenshots**
[image]
[image]
[image]

AlexandraTrifan, 2024-12-11:
Hi!

The initial release candidate for the new Scopy v2.0.0 is now available in the official releases page: https://github.com/analogdevicesinc/scopy/releases/tag/v2.0.0-beta-rc1. This candidate includes a lot of new features in Scopy while bringing multiple bug fixes for the ADALM2000 plugin and underlying libm2k library.

Could you give this version a try and reopen the issue if the bug still appears?

Thank you!

closed 2024-12-11 by AlexandraTrifan

lysynski, 2024-12-13:
Hi

Thanks for the link. I downloaded the MacOS version, I see there’s more than a DMG in the zip file but there’s not instructions for installation. Should I follow typical installation and just copy Scopy from the DMG to my applications folder or is there more that needs to be done?

Thanks
Boris

lysynski, 2024-12-17:
> Hi!
> 
> The initial release candidate for the new Scopy v2.0.0 is now available in the official releases page: https://github.com/analogdevicesinc/scopy/releases/tag/v2.0.0-beta-rc1. This candidate includes a lot of new features in Scopy while bringing multiple bug fixes for the ADALM2000 plugin and underlying libm2k library.
> 
> The pull request regarding this issue is: #1496 Could you give this version a try and reopen the issue if the bug still appears?
> 
> Thank you!

Hi

Thanks for the link. I downloaded the MacOS version, I see there’s more than a DMG in the zip file but there’s not instructions for installation. Should I follow typical installation and just copy Scopy from the DMG to my applications folder or is there more that needs to be done?

Thanks
Boris

adisuciu, 2024-12-17:
Yes, typical installation of a DMG file. The rest of the files are auxiliary to the build process, used mainly for debugging. We'll clean them in a future version.

lysynski, 2024-12-18:
I tried the new release... you need more testing.
I can enable channels and group them but the visual for this needs to be more obvious as the white frame around the channels isn't obvious and it's difficult to see that they're part of the group. Once grouped, I switch the pattern to SPI. It's not obvious which channel is what signal. I enabled ch0-2, ch0 became clk, ch1 data, ch2 enable; there's no indicator to show this and I have no control to change it. Changing the SPI parameters (buttons) doesn't update the generated pattern shown on screen even if I run the pattern generator. Output is also unaffected. It only changes if I modify one of the data fields.

AlexandraTrifan, 2024-12-20:
From a UI/UX point of view, the M2K plugin needs a rework. We've been focused on the new Scopy infrastructure in order to provide support for other boards. This rework of the M2k plugin would include a different way to display channel information for grouped channels, information about their roles in the decoder - maybe similar to the one found in the Logic Analyzer. 
For now, the channel order in the decoder can be changed by grabbing the Burger Menu Button for each channel in the Decoder Menu (right side menu) and dragging it above or below other channels. 

As for the last remark, the buttons don't apply indeed. It seems to be a side effect of a major change done in the last month. I've added a fix for this (https://github.com/analogdevicesinc/scopy/pull/1867) and it will be included in the next release candidate.

Thank you!

lysynski, 2024-12-21:
I tired moving the signals with the burger menu. After a couple moves they lose their positioning in the list, shifted down with nothing above. I tried to put them back in order and the app crashed.