analogdevicesinc/scopy #1526: Oscilloscope: After drawing cursors, pressing Ctrl+Z and/or Escape should reset the view

DeflateAwning, 2023-12-23:
**Is your feature request related to a problem? Please describe.**
I'm frustrated that I can't figure out how to reset the view after drawing cursors! 

**Describe the solution you'd like**
I want all of the following to reset to my last view before I drew the cursors:
1. Pressing Ctrl+Z
2. Pressing Escape
3. Clicking a large button that appears to the left of "Print" and "Enable Mixed Signal View". This button should say "<left arrow> Undo Cursor Zoom"
4. Back button on a mouse with forwards/back navigation buttons.

**Describe alternatives you've considered**
I still can't figure out how you're supposed to get back. It almost feels like a bug. 

Using Linux, v1.4.1.

AlexandraTrifan, 2024-12-11:
Hi,

By drawing cursors you mean drawing the zoom view i presume. 
In order to return to the previous zoom level you can use the mouse right click. In order to reset all zoom levels, Escape should work. 
As stated in other recently closed or commented issues, we suggest you try the latest Scopy release candidate Scopy v2.0.0. This update significantly improves the user experience.

Thank you!

closed 2024-12-11 by AlexandraTrifan

DeflateAwning, 2024-12-11:
It would have been nice if some of these issues were considered and consoluted during the development of v2, but I'm very excited to try it! Thank you!