analogdevicesinc/scopy #1527: Oscilloscope: add option to trigger off of Logic DIO when in Mixed Signal Mode

DeflateAwning, 2023-12-23:
**Is your feature request related to a problem? Please describe.**
I wish I could trigger off of a DIO rising edge or similar when in Mixed Signal Mode (i.e., showing both analog and DIO on the screen).

An example use case is where I use a digital signal to trigger an analog sensor with two analog output channels. I want to view the two analog output channels with the two analog pins (on a ADALM2000, for example), but trigger it off the DIO signal to the analog sensors. This configuration is helpful for cases where I want to confirm that the analog sensors respond with a constant low signal when queried via the DIO pin.

**Describe the solution you'd like**
In the Triggers pane (see screenhot, very right side), I want to be able to select DIO0 in the "Source" dropdown.

**Describe alternatives you've considered**
Currently, in the problem I described above, I have to probe the two analog channels separately, using one analog scope channel as the trigger signal. 

**Additional**
Even if this is not implemented, can you please confirm if this is a hardware limitation, or just a lack of implementation in the current Scopy software? If it is simply a Scopy limitation, I may take a stab at implementing the feature; however, if it is physically impossible with the given hardware, please tell me so I don't waste time trying.

AlexandraTrifan, 2024-12-11:
Hi,

If i understand correctly, you would like to trigger on a condition of DIO0, without triggering on analog CH1 or CH2. Please correct me if I am wrong. 
In that case, in the Trigger settings menu, you can disable the "ANALOG" trigger and use the Logic Analyzer trigger configuration to control which DIO channels will trigger the Mixed Signal View. (please see the attached screenshots)
[image]

[image]

DeflateAwning, 2024-12-11:
Neat! That looks like exactly what I was after back then.

Was this possible in v1.x? Like, did I just miss it, or is this a new feature?

AlexandraTrifan, 2024-12-12:
I think this was also available in previous versions.