analogdevicesinc/scopy #1531: Home: Scopy exits or hangs after connect in calibration

Modiug, 2024-01-19:
**Environment:**
 - OS: Windows 10 Enterprise, 22H2, 19045.3930
 - FW Version:  v0.32
 - SW Version or git tag: Scopy v1.4.1

**Describe the bug**
After clicking connect, "Connecting" and then "Calibrating" is displayed. After a few seconds the Scopy window closes, or it hangs forever. This only happens on my work laptop (Lenovo ThinkPad P15 Gen 1). No problems on another laptop under both, Windows 10 and Linux (LMDE 6).

**To Reproduce**
Steps to reproduce the behavior:
As described above
1. Select device
2. Click on 'Connect'
3. Window closes after a few seconds or hangs with a busy indicator

**Expected behavior**
Finish calibration after a few seconds and resume normal operation

**Additional context**
I tried the usual:
- Cleaning the AppData\Roaming\ADI folder.
- Clicking "Reset Scopy" in Preferences.

Other observations:
- My work laptop is managed by corporate IT and uses Windows EPM (end point management)
- Interestingly GnuRadio with the M2K driver DOES work on this machine (even with "Calibrate ADC: True")
- I have two M2Ks, both show the same behavior with Scopy on my work laptop

Unfortunately, I cannot recall if this ever worked on this machine. Scopy had worked on a previous machine. It might be related to a Windows update or the way how priviledges are managed (EPM?).

AlexandraTrifan, 2024-01-19:
Hi,

Before trying to connect to the M2K - could you check in the Preferences page for a setting related to OpenGL and if checked, uncheck it. Then try to connect to the M2K and see if this still happens?
If it still does, we can investigate further.

Thank you!
-Alexandra

Modiug, 2024-01-22:
Hi Alexandra,

no, unfortunatly unchecking "use hardware accelerated plotting - OpenGL (EXPERIMENTAL)" did not work.
I tried the Flatpak on Ubuntu 20.04 running inside Hyper-V on the same machine and it works. But here I connected via IP instead of usb directly.
On Windows USB is used by default, so I tried IP as well, but no change. The Scopy application still exits or hangs after calibration.

With session logging enabled I do not get a crashdump, only the log file. Nothing spectecular there. It ends with:
```
I20240122 08:42:09.931924 22552 contextbuilder.cpp:206] libm2k version: v0.6.0
I20240122 08:42:09.931924 22552 contextbuilder.cpp:211] libiio version: 0.25
I20240122 08:42:09.931924 22552 contextbuilder.cpp:216] libusb version: 1.0.26.11724
I20240122 08:42:09.931924 22552 contextbuilder.cpp:221] Firmware version: v0.32
I20240122 08:42:10.016561 22552 contextbuilder.cpp:206] libm2k version: v0.6.0
I20240122 08:42:10.016561 22552 contextbuilder.cpp:211] libiio version: 0.25
I20240122 08:42:10.016561 22552 contextbuilder.cpp:216] libusb version: 1.0.26.11724
I20240122 08:42:10.016561 22552 contextbuilder.cpp:221] Firmware version: v0.32
I20240122 08:42:10.016561 22340 channel.cpp:316] [m2k-adc][voltage0][calibbias][read] 2034
I20240122 08:42:10.079075 22552 channel.cpp:273] [ad9963][temp0][raw][read] 1648.000000
I20240122 08:42:10.079075 22552 channel.cpp:273] [ad9963][temp0][offset][read] -1366.000000
I20240122 08:42:10.079075 22552 channel.cpp:273] [ad9963][temp0][scale][read] 200.000000
I20240122 08:42:10.079075 22552 contextbuilder.cpp:206] libm2k version: v0.6.0
I20240122 08:42:10.079075 22552 contextbuilder.cpp:211] libiio version: 0.25
I20240122 08:42:10.079075 22552 contextbuilder.cpp:216] libusb version: 1.0.26.11724
I20240122 08:42:10.079075 22552 contextbuilder.cpp:221] Firmware version: v0.32
I20240122 08:42:10.079075 22552 devicegeneric.cpp:200] [m2k-adc][sampling_frequency][read] 100000000.000000
I20240122 08:42:10.101227 22552 channel.cpp:330] [m2k-adc-trigger][voltage5][out_direction][write] out
I20240122 08:42:10.101227 22552 channel.cpp:330] [m2k-adc-trigger][voltage5][out_select][write] sw-trigger
I20240122 08:42:10.163733 22552 devicegeneric.cpp:475] [m2k-adc][sampling_frequency_available][read] 1000 10000 100000 1000000 10000000 100000000
```

Any other suggestions what I could test?

Kind regards
Guido

Modiug, 2025-01-06:
Sorry for the late reply. The v2.0.0-beta-rc1 solves the issue for me.
Thanks a lot!