analogdevicesinc/scopy #1551: Spikes in the plots by Scopy Network Analyser

SteveNg2013, 2024-02-22:
**Environment:**
 - OS: MacOS 13.5.2
 - FW Version 0.32
 - SW Version Scopy - v1.2.0 - e02cebe

**Describe the bug**
When running the Network Analyser I notice spikes on the magnitude and phase plots. This is similar to bug reported in 2017 (Spikes appear on the Network Analyzer plot #167) but using a very simple circuit.

**To Reproduce**

I wired up a simple bandstop filter with a 1k resistor  in series with a DIY 288uH toroid inductor and a 0.1uF ceramic cap all in series and set the analyser to sweep from 10kHz to 100kHz and taking 100 samples. Amplitude was set to 10V.

Steps to reproduce the behavior:
Wire the filter.

1+, W1.            2+                             1-, 2-, GND
---------/\/\/\------()()()------||------
                 1k.         288uH.       0.1uF

**Expected behavior**
I was expecting a smooth plot of magnitude and phase but get random spikes on both plots.

**Screenshots**
A screenshot is included

**Additional context**

The ini files might be useful to reproduce the error.
The ini file is located at: C:/Users/ali/AppData/Roaming/ADI/Scopy.ini
Please consider attaching it.

WHERE CAN I FIND THE INI FILE IN MACOS?

Add any other context about the problem here.
[Screenshot 2024-02-22 at 4.56.46 pm.pdf](https://github.com/analogdevicesinc/scopy/files/14369800/Screenshot.2024-02-22.at.4.56.46.pm.pdf)

adisuciu, 2024-02-22:
Is it possible to update to the latest Scopy v1.4.1 ? 
If the bug still persists, you can use the network analyzer buffer previewer functionality that shows the time domain waveform for different samples when acquired.

SteveNg2013, 2024-02-22:
I updated to 1.4.1 and still found the spikes. I am attaching a screen grab with the buffer preview. A look at the buffer at the point of the spikes shows nothing captured in the buffer. Is my interpretation correct? Question obviously is why the hiccup which occurs quite randomly?

[Screenshot 2024-02-22 at 9.49.00 pm.pdf](https://github.com/analogdevicesinc/scopy/files/14372848/Screenshot.2024-02-22.at.9.49.00.pm.pdf)

SteveNg2013, 2024-02-22:
Adrian,

Thanks for taking a look at the problem and your suggestion. I also wanted to add that I ran the same simulation in a Linux Mint 20 virtual machine on the same iMac and saw the same randomly occurring spikes.

adisuciu, 2024-02-22:
There are 2 waveforms in the buffer. 
The stimulus and the response. 

There is no stimulus, and therefore there is no response ... the acquisition is correct though - you are seeing a (probably) floating signal around 0 volts

It almost looks like the circuit/connections have spurious failures - are you using a breadboard ? Can you confirm you are getting the same spikes when running the network analyzer in a loop (without a device under test) .. simply loop W1 to 1+ and W2 to 2+. You could also try different connection wires.

SteveNg2013, 2024-02-22:
Thanks.  I will do it tomorrow and let you know. Just in case you’re
wondering, I’m in Sydney Australia.

SteveNg2013, 2024-02-23:
Adrian,I tried looping directly as you suggested, the same random spikes occurred. Yes, I am using a breadboard. I even changed breadboards but the same problem occurred. As for the wires, apart from the connectors that came with the ADALM2000 which I can’t do much about now, I’ve tried different wires and the problem did not go away. I wonder if there is a problem with either the ADC or the DAC in the ADALM2000.Thanks.Sent from my iPad

adisuciu, 2024-02-27:
We've tracked this down to the firmware version. Can you confirm that downgrading to firmware v0.31 fixes the problem ?

SteveNg2013, 2024-02-27:
Hello Adrian,

Thanks for tracking down the problem. I downgraded to v0.31 and can confirm that I have not seen any spikes in the traces. Thanks again.

adisuciu, 2024-02-27:
Thanks, we're looking into a fix in the firmware ..

closed 2024-09-24 by adisuciu