analogdevicesinc/scopy #1568: Scopy crashes after clicking on Channel Settings in Oscilloscope

hrayl, 2024-03-17:
**Environment:**
 - OS: Windows 11
 - FW Version: v0.25
 - SW Version or git tag: v1.4.1 - f4beeb1

**Bug Description:**
Scopy crashes whenever I click on the settings icon for CH1 and CH2 in Oscilloscope (sometimes for CH1 and CH2 in Signal Generator too). 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Oscilloscope'
2. Click on either Channel Settings for CH1 or CH2.
3. Scopy crashes.

**Expected behavior**
Go into the channel settings menu/Programme not crashing.

**Screenshots**
My screen after clicking on CH2 Channel Setting. It just freezes there.
[image]

Benlexhi, 2024-11-13:
I am having the same issue. Was there any fix?

AlexandraTrifan, 2024-12-11:
Hi!

The initial release candidate for the new Scopy v2.0.0 is now available in the official releases page: https://github.com/analogdevicesinc/scopy/releases/tag/v2.0.0-beta-rc1 .

Could you give this version a try and reopen the issue if the bug still appears? We ran a test with this version on a Windows 11 and couldn't reproduce it.

Thank you!

closed 2024-12-11 by AlexandraTrifan