analogdevicesinc/scopy #1771: ADALM2000: Crashes immediately after attempting to calibrate

butterflyhigh, 2024-10-16:
**Environment:**
 - OS: 6.10.10-arch1-1 (Linux)
 - FW: 0.27
 - SW: 1.4.1

**Describe the bug**
When I attempt to calibrate the ADALM2000 by clicking the calibrate button, Scopy crashes immediately. If started from the terminal (via flatpak run), I get this output:

```
QIODevice::read (QFile, ":/stylesheets/themes/global.qss"): device not open
QIODevice::read (QFile, ":stylesheets/stylesheets/browserStylesheet.qss"): device not open
Warning: Using function expressions as statements in scripts is not compliant with the ECMAScript specification:
 "function(o) { for (each in o) { print(each); } }..." 
This will throw a syntax error in Qt 5.12. If you want a function expression, surround it by parentheses.
Unknown property link-color
Unknown property link-color
qt.qpa.wayland: Wayland does not support QWindow::requestActivate()
ERROR: Unable to claim interface 1:8:5: Device or resource busy (16)
E20241016 11:34:04.782114    11 devicegeneric.cpp:650] Device: Cannot set the number of kernel buffers
```
However, this is all printed before crashing. Nothing new is printed when I click calibrate.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect ADALM2000
2. Run Scopy
3. Connect to ADALM2000
4. Click calibrate

I am not familiar with this software or the ADALM2000, this is my first time using it. This is a fresh install of Scopy without any changed configuration. I would like to know how to either not experience this or work around it, since I need it for an assignment due tomorrow.

AlexandraTrifan, 2024-10-17:
Could you try to update the firmware on your M2K?

butterflyhigh, 2024-10-17:
> Could you try to update the firmware on your M2K?

I am able to calibrate with the latest firmware, thank you. Could I maybe request adding a note on this issue somewhere in the wiki or README, or possibly somewhere in the code to handle this error instead of segfaulting without explanation? My M2k is very new (purchased in September), so I didn't think of outdated firmware as a potential issue at first. I also figure I'm not the only one who had this issue and spent a long time trying to fix it, and I probably would've fixed it a lot earlier (and finished my assignment earlier LOL) if I had known updating the firmware would resolve it. Once I realized the issue I saw that this is a known issue, but it was very difficult to find information when all I knew was that it segfaulted. 

Either way, I can use Scopy now, I just think it might be nice to have more available info on this. Cheers

closed 2024-10-18 by adisuciu