analogdevicesinc/scopy #1857: Scopy v2.0.0 Linux x86_64 AppImage segfaults

granquet, 2024-12-13:
**Environment:**
 - OS: Gentoo
 - v2.0.0 beta rc1 AppImage

**Describe the bug**
AppImage segfault when starting

**To Reproduce**
Steps to reproduce the behavior:
1. Run the AppImage
2. Admire segfault

**Additional context**

```
➜  Downloads ./Scopy-v2.0.0-beta-rc1-Linux-x86_64.AppImage
Starting Scopy
        linux-vdso.so.1 (0x00007f524551d000)
        libscopy-core.so => /tmp/.mount_Scopy-VhOu3b/usr/lib/libscopy-core.so (0x00007f5245242000)
        libscopy-common.so => /tmp/.mount_Scopy-VhOu3b/usr/lib/libscopy-common.so (0x00007f5245239000)
        libQt5Widgets.so.5 => /tmp/.mount_Scopy-VhOu3b/usr/lib/libQt5Widgets.so.5 (0x00007f5244600000)
        libQt5Gui.so.5 => /tmp/.mount_Scopy-VhOu3b/usr/lib/libQt5Gui.so.5 (0x00007f5243a00000)
        libQt5Core.so.5 => /tmp/.mount_Scopy-VhOu3b/usr/lib/libQt5Core.so.5 (0x00007f5243000000)
        libstdc++.so.6 => /usr/lib/gcc/x86_64-pc-linux-gnu/14/libstdc++.so.6 (0x00007f5242c00000)
        libgcc_s.so.1 => /usr/lib/gcc/x86_64-pc-linux-gnu/14/libgcc_s.so.1 (0x00007f52451e7000)
        libc.so.6 => /lib64/libc.so.6 (0x00007f524501f000)
        libscopy-gui.so => /tmp/.mount_Scopy-VhOu3b/usr/lib/libscopy-gui.so (0x00007f5242800000)
        libscopy-pluginbase.so => /tmp/.mount_Scopy-VhOu3b/usr/lib/libscopy-pluginbase.so (0x00007f5244fd8000)
        libscopy-iioutil.so => /tmp/.mount_Scopy-VhOu3b/usr/lib/libscopy-iioutil.so (0x00007f5244fa2000)
        libsigrokdecode.so.4 => /tmp/.mount_Scopy-VhOu3b/usr/lib/libsigrokdecode.so.4 (0x00007f5244f8b000)
        libQt5Concurrent.so.5 => /tmp/.mount_Scopy-VhOu3b/usr/lib/libQt5Concurrent.so.5 (0x00007f5242200000)
        libQt5Qml.so.5 => /tmp/.mount_Scopy-VhOu3b/usr/lib/libQt5Qml.so.5 (0x00007f5241800000)
        libiio.so.0 => /tmp/.mount_Scopy-VhOu3b/usr/lib/libiio.so.0 (0x00007f5244f64000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f524551f000)
        libpthread.so.0 => /lib64/libpthread.so.0 (0x00007f5244f5f000)
        libGL.so.1 => /usr/lib64/libGL.so.1 (0x00007f5244eea000)
        libm.so.6 => /lib64/libm.so.6 (0x00007f5244555000)
        libz.so.1 => /usr/lib64/libz.so.1 (0x00007f5244ed0000)
        libicui18n.so.56 => /tmp/.mount_Scopy-VhOu3b/usr/lib/libicui18n.so.56 (0x00007f5241000000)
        libicuuc.so.56 => /tmp/.mount_Scopy-VhOu3b/usr/lib/libicuuc.so.56 (0x00007f5240a00000)
        libicudata.so.56 => /tmp/.mount_Scopy-VhOu3b/usr/lib/libicudata.so.56 (0x00007f523ee00000)
        libdl.so.2 => /lib64/libdl.so.2 (0x00007f5244ec9000)
        libgthread-2.0.so.0 => /tmp/.mount_Scopy-VhOu3b/usr/lib/libgthread-2.0.so.0 (0x00007f5244ec3000)
        libglib-2.0.so.0 => /tmp/.mount_Scopy-VhOu3b/usr/lib/libglib-2.0.so.0 (0x00007f524441e000)
        libQt5Xml.so.5 => /tmp/.mount_Scopy-VhOu3b/usr/lib/libQt5Xml.so.5 (0x00007f523e800000)
        libQt5Svg.so.5 => /tmp/.mount_Scopy-VhOu3b/usr/lib/libQt5Svg.so.5 (0x00007f523e200000)
        libqwt.so.6.4 => /tmp/.mount_Scopy-VhOu3b/usr/lib/libqwt.so.6.4 (0x00007f5243831000)
        libQt5Network.so.5 => /tmp/.mount_Scopy-VhOu3b/usr/lib/libQt5Network.so.5 (0x00007f523dc00000)
        libserialport.so.0 => /tmp/.mount_Scopy-VhOu3b/usr/lib/libserialport.so.0 (0x00007f5244ead000)
        libpython3.8.so.1.0 => /tmp/.mount_Scopy-VhOu3b/usr/lib/libpython3.8.so.1.0 (0x00007f523d600000)
        librt.so.1 => /lib64/librt.so.1 (0x00007f5244ea6000)
        libusb-1.0.so.0 => /usr/lib64/libusb-1.0.so.0 (0x00007f5244e87000)
        libavahi-client.so.3 => /tmp/.mount_Scopy-VhOu3b/usr/lib/libavahi-client.so.3 (0x00007f5244e72000)
        libavahi-common.so.3 => /tmp/.mount_Scopy-VhOu3b/usr/lib/libavahi-common.so.3 (0x00007f5244e62000)
        libxml2.so.2 => /tmp/.mount_Scopy-VhOu3b/usr/lib/libxml2.so.2 (0x00007f5242639000)
        libGLdispatch.so.0 => /usr/lib64/libGLdispatch.so.0 (0x00007f52443a2000)
        libGLX.so.0 => /usr/lib64/libGLX.so.0 (0x00007f524436f000)
        libpcre.so.3 => /tmp/.mount_Scopy-VhOu3b/usr/lib/libpcre.so.3 (0x00007f5242f8c000)
        libQt5OpenGL.so.5 => /tmp/.mount_Scopy-VhOu3b/usr/lib/libQt5OpenGL.so.5 (0x00007f523d000000)
        libgssapi_krb5.so.2 => /tmp/.mount_Scopy-VhOu3b/usr/lib/libgssapi_krb5.so.2 (0x00007f5242f3c000)
        libexpat.so.1 => /usr/lib64/libexpat.so.1 (0x00007f5244344000)
        libutil.so.1 => /lib64/libutil.so.1 (0x00007f524433f000)
        libdbus-1.so.3 => /tmp/.mount_Scopy-VhOu3b/usr/lib/libdbus-1.so.3 (0x00007f5242ee5000)
        libicuuc.so.66 => /tmp/.mount_Scopy-VhOu3b/usr/lib/libicuuc.so.66 (0x00007f523cc00000)
        liblzma.so.5 => /tmp/.mount_Scopy-VhOu3b/usr/lib/liblzma.so.5 (0x00007f5243806000)
        libX11.so.6 => /usr/lib64/libX11.so.6 (0x00007f52424f5000)
        libkrb5.so.3 => /tmp/.mount_Scopy-VhOu3b/usr/lib/libkrb5.so.3 (0x00007f5242411000)
        libk5crypto.so.3 => /tmp/.mount_Scopy-VhOu3b/usr/lib/libk5crypto.so.3 (0x00007f5242eb2000)
        libcom_err.so.2 => /usr/lib64/libcom_err.so.2 (0x00007f5244336000)
        libkrb5support.so.0 => /tmp/.mount_Scopy-VhOu3b/usr/lib/libkrb5support.so.0 (0x00007f5242ea1000)
[... 52 lines cut ...]
[ I Benchmark 16:26:47.991 ] file:///__w/scopy/scopy/core/src/scopymainwindow.cpp:434 - OpenGL load status:  true
[ scopy::CrashReport::signalHandler
?libc.so.6?
?libxdg-shell.so?
QtWaylandClient::QWaylandXdgShellIntegration::initialize
QtWaylandClient::QWaylandShellIntegrationFactory::create
QtWaylandClient::QWaylandIntegration::createShellIntegration
QtWaylandClient::QWaylandIntegration::initializeShellIntegration
QtWaylandClient::QWaylandIntegration::shellIntegration
QtWaylandClient::QWaylandWindow::shouldCreateShellSurface
QtWaylandClient::QWaylandWindow::initWindow
QtWaylandClient::QWaylandWindow::setVisible
QWindowPrivate::setVisible
QWidgetPrivate::show_sys
QWidgetPrivate::show_helper
QWidgetPrivate::setVisible
main
?libc.so.6?
__libc_start_main
_start 16:26:47.998 ]  - Scopy finished with error

[1]    19734 segmentation fault  ./Scopy-v2.0.0-beta-rc1-Linux-x86_64.AppImage
```

Note that v1.5.0-rc1 also segfaults for me.

adisuciu, 2024-12-13:
Can you try running it like this ? 

`QT_QPA_PLATFORM=xcb ./Scopy-v2.0.0-beta-rc1-Linux-x86_64.AppImage`

granquet, 2024-12-13:
> Can you try running it like this ?
> 
> `QT_QPA_PLATFORM=xcb ./Scopy-v2.0.0-beta-rc1-Linux-x86_64.AppImage`

This works, thank you.

I've toyed with this and running with `QT_QPA_PLATFORM` = `eglfs` or `xcb` works, but not `wayland` which reproduces the crash.

Anything I can do on my end to "fix" the wayland plugin?

adisuciu, 2024-12-17:
I think this may be broken from our end. We're looking into it.

bindea-cristian, 2025-03-19:
There was a missing .so in the AppImage.
This was fixed in this PR https://github.com/analogdevicesinc/scopy/pull/1972, specifically by this commit https://github.com/analogdevicesinc/scopy/commit/ba22ba2d5d1f1067bd11f8eadc76d86fb81e2d23.

closed 2025-03-19 by bindea-cristian