analogdevicesinc/scopy #1904: Emulator: AppImage emulator not working

Andrei-Fabian-Pop, 2025-02-13:
**Environment:**
 - OS: Linux 
 - Git hash: 3428b10

**Describe the bug**
The linux AppImage does not correctly load the emulator. Currently, the only option to get is working is to go to 'Preferences', scroll down and click 'Reset' and restart Scopy. This needs to be done every time Scopy is started (and not restarted). This happens because an AppImage is mounted on a different path every time it opens.

**To Reproduce**
Steps to reproduce the behavior:
1. Open Scopy
2. Click on 'Emulator'
3. See ''Can't find iio-emu in the system!" error

**Expected behavior**
The iio-emu should be found without restarting Scopy.

andrei47w, 2025-02-25:
On windows 10, even if the path in preferences.ini is bad and iio-emu is not installed in system, emu will be detected and work in Scopy as long as iio-emu.exe is next to scopy.exe. Is this intended?

Andrei-Fabian-Pop, 2025-02-25:
Yes

closed 2025-03-06 by AlexandraTrifan