analogdevicesinc/scopy #1973: m2k: Two Channel DAC Output Crash

erfan-khadem, 2025-03-14:
**Environment:**
 - OS: Linux (Arch, Gnome Wayland, Everything up-to-date)
 - Latest
 - 2.0.0.rc1 and a recent CI artifact (3ddf222f, ~March 8th)

**Bug Description**
I am trying to draw a picture on an analog oscilloscope. I generate the XY coordinates of the said picture using a python script (as a .csv files with two columns for both channels). When I try to load the CSV file in the waveform generation plugin (using the buffer feature) *sometimes* the program crashes with a log line like:

`Expected a positive value for interpolation but got 0 instead.`

I don't have access to my hardware to reproduce the issue right now or get coredumps but I will try my best to provide details and reproduction steps.

**To Reproduce**
- Use the attached CSV file
- Start the waveform generator with a 1KHz sinewave on both channels (might be irrelevant)
- Select the CSV files as the buffer. Use column 0 for channel 1 and column 1 for channel 2
- Change the sample rate from 10KSPS or something similar straight to 5MSPS
- If the crash does not happen immediately, try changing the sample rate for the other channel
- If it still does not crash, load / reload other CSV files and change the sample rate

Another curious point is that if I first change the sample rate from 10KSPS to 100K, then 1Meg, then 5Meg I believe that the program never crashes. Or if I only change the sample rate in the MSPS range (and not go below) still I have no issues. Another thing to mention is that I ~reliably get a crash if I change the sample rate by using the dropdown menu to change the multiplier.

I am able to reproduce this using both the flatpak version and the AppImage version.

[output.csv](https://github.com/user-attachments/files/19255439/output.csv)
[another.csv](https://github.com/user-attachments/files/19255440/dr-kaboli.csv)

AlexandraTrifan, 2025-03-17:
Hi, 

We'll investigate and come back as soon as possible.

Thank you!

erfan-khadem, 2025-03-22:
Does c073f467666aba9a723d280e718b4e342267a103 fix this issue? If so feel free to close this issue (I will be able to test using hardware on April 10th)

AlexandraTrifan, 2025-03-24:
Yes, the linked commit should fix this issue. 
I am closing this issue. Can be reopened if the problem persists.

closed 2025-03-24 by AlexandraTrifan