analogdevicesinc/scopy #1994: <Instrument>: <Title>

tfcollins, 2025-04-03:
**Environment:**
 - OS: Ubuntu 22.04 (i3 WM)
 - FW Version: NA
 - SW Version: 2.0.0-b3d077f6 (2.0 release) Flatpak

**Describe the bug**
No icon images

**To Reproduce**
Steps to reproduce the behavior:
1. Open scopy

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

**Additional context**

adisuciu, 2025-04-07:
This may be due to a clash between old preferences and new preferences. We're adding a default behavior.

andrei47w, 2025-04-07:
Can be fixed by going to preferences and resetting the app to default settings? I have will push a fix for this

closed 2025-04-14 by adisuciu