analogdevicesinc/scopy #1995: No axis labels Time or Frequency Plots

tfcollins, 2025-04-03:
**Environment:**
 - OS: Ubuntu 22.04
 - FW Version : PlutoSDR v0.38
 - SW Version or git tag: Release 2.0 Flatpak

**Describe the bug**
There are no y or x axis labels in Time or Frequency Plot

**To Reproduce**
Steps to reproduce the behavior:
1. Connect PlutoSDR
2. Launch Scopy
3. Start ADC Time or ADC Frequency plugins

**Expected behavior**
Any plot in engineering should have units

**Screenshots**

[image]

adisuciu, 2025-04-04:
I guess it's a little hidden or it should be on by default but you can activate plot labels from the plot settings menu 

[image]

tfcollins, 2025-04-09:
Yeah should be on by default. Also, we will need to add better units than dB. For most cases it should be dBFS unless you have a calibrated device