analogdevicesinc/scopy #1996: ADALM2000: Modules not Found on MacOS 15.4

jhannemann, 2025-04-05:
**Environment:**
 - OS: MacOS 15.4 on M1 Max
 - FW Version 0.33
 - SW Version 2.0.0

**Describe the bug**
When starting Scopy, the only plugin that is found and shown is the DACPlugin

**To Reproduce**
Steps to reproduce the behavior:
1. Open Scopy
2. A scan only shows the DACplugin with USB 1.2.5
3. Click "+"
4. A scan will show "Analog Devices M2k (ADALM 2000) [usb 1.2.5]
5. Click "Verify"
6. It will show firmware version 0.33 and M2k Rev.D (Z7010)
7. Compatible plugins are shown: DAC, DataLogger, Debugger, Regmap
8. No other plugins are shown
9. Click on Add Device
10. Another DACPlugin shows up in the device list, but no ADALM.

**Expected behavior**
ADALM2000 should show up as a device

**Additional context**

No ini files are present.

bindea-cristian, 2025-04-11:
The problem was that some of the dylibs bundled inside the installer contained duplicated LC_RPATHs.
I've fixed the issue and updated the installer from the release. Can you check the app again ?
https://github.com/analogdevicesinc/scopy/releases/download/v2.0.0/Scopy-v2.0.0-osx_15.2.zip

jhannemann, 2025-04-11:
Works like a charm! Thank you!

closed 2025-04-11 by bindea-cristian