analogdevicesinc/scopy #2016: Multiple Tools: No Axis Labels in Graphs

HTCPCP-Error418, 2025-05-01:
**Environment:**
 - OS: Windows
 - FW Version: v0.33
 - SW Version: 2.0.0-b3d077f6

**Describe the bug**
In the screenshots that exist in the documentation, the graphs are clearly labeled with units (such as volts and ms). These labels do not appear in my version of the tools. After creating this bug report, I also noticed that there are no axis labels in the Oscilloscope tool.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Signal Generator'
2. Create any signal (constant, waveform, buffer, or math)
3. Observe missing unit labels

**Expected behavior**
I expect the values on the graph axis to have unit labels (or at least one label on each axis)

**Screenshots**

**Additional context**

The ini files might be useful to reproduce the error.
The ini file is located at: C:/Users/ali/AppData/Roaming/ADI/Scopy.ini
Please consider attaching it.

Add any other context about the problem here.

I searched through any settings menus I could find and was unable to find any setting to change this.

andrei47w, 2025-05-08:
To see plot labels in the M2K plugin you need to go in Preferences, M2kPlugin and under the Analog tools panel "Enable labels on the plot" can be enabled.
If you are using the ADC plugin you can go to either the Time or Frequency tool and within the plot settings there should be a "Show plot labels" checkbox.

HTCPCP-Error418, 2025-05-14:
I double-checked that I had "Enable labels on the plot" turned on and it still appears as though the plots in Signal Generator do not have labels on either axis. I do see that the Oscilloscope tool has the time per division above the graph on the top-left side, though.

[image]

[image]

[image]

closed 2025-05-28 by AlexandraTrifan