analogdevicesinc/scopy #2026: Fails to build `axishandle.cpp.o` with `axishandle.cpp:59:44: error: request for member ‘isXAxis’ in ‘((scopy::AxisHandle*)this)->scopy::AxisHandle::m_axisId’, which is of non-class type ‘QwtAxisId’ {aka ‘int’}`.

dreirund, 2025-05-14:
I am trying the latest git checkout from this on Artix GNU/Linux (rolling release).

Build of `axishandle.cpp.o` fails with  
`axishandle.cpp:59:44: error: request for member ‘isXAxis’ in ‘((scopy::AxisHandle*)this)->scopy::AxisHandle::m_axisId’, which is of non-class type ‘QwtAxisId’ {aka ‘int’}`  
and subsequent errors:  
```
[  6%] Building CXX object gui/CMakeFiles/scopy-gui.dir/src/axishandle.cpp.o
/var/cache/makepkg/build/scopy-git/src/scopy/gui/src/axishandle.cpp: In member function ‘int scopy::AxisHandle::getPos()’:
/var/cache/makepkg/build/scopy-git/src/scopy/gui/src/axishandle.cpp:59:44: error: request for member ‘isXAxis’ in ‘((scopy::AxisHandle*)this)->scopy::AxisHandle::m_axisId’, which is of non-class type ‘QwtAxisId’ {aka ‘int’}
   59 | int AxisHandle::getPos() { return m_axisId.isXAxis() ? m_pos.x() : m_pos.y(); }
      |                                            ^~~~~~~
/var/cache/makepkg/build/scopy-git/src/scopy/gui/src/axishandle.cpp: In member function ‘void scopy::AxisHandle::setPosSilent(int)’:
/var/cache/makepkg/build/scopy-git/src/scopy/gui/src/axishandle.cpp:61:51: error: request for member ‘isXAxis’ in ‘((scopy::AxisHandle*)this)->scopy::AxisHandle::m_axisId’, which is of non-class type ‘QwtAxisId’ {aka ‘int’}
   61 | void AxisHandle::setPosSilent(int pos) { m_axisId.isXAxis() ? m_pos.setX(pos) : m_pos.setY(pos); }
      |                                                   ^~~~~~~
/var/cache/makepkg/build/scopy-git/src/scopy/gui/src/axishandle.cpp: In member function ‘void scopy::AxisHandle::setHandle(scopy::HandleOrientation)’:
/var/cache/makepkg/build/scopy-git/src/scopy/gui/src/axishandle.cpp:93:21: error: request for member ‘isYAxis’ in ‘((scopy::AxisHandle*)this)->scopy::AxisHandle::m_axisId’, which is of non-class type ‘QwtAxisId’ {aka ‘int’}
   93 |         if(m_axisId.isYAxis())
      |                     ^~~~~~~
/var/cache/makepkg/build/scopy-git/src/scopy/gui/src/axishandle.cpp:104:29: error: request for member ‘isYAxis’ in ‘((scopy::AxisHandle*)this)->scopy::AxisHandle::m_axisId’, which is of non-class type ‘QwtAxisId’ {aka ‘int’}
  104 |                 if(m_axisId.isYAxis()) {
      |                             ^~~~~~~
/var/cache/makepkg/build/scopy-git/src/scopy/gui/src/axishandle.cpp: In member function ‘QRect scopy::AxisHandle::getBigRect()’:
/var/cache/makepkg/build/scopy-git/src/scopy/gui/src/axishandle.cpp:144:21: error: request for member ‘isXAxis’ in ‘((scopy::AxisHandle*)this)->scopy::AxisHandle::m_axisId’, which is of non-class type ‘QwtAxisId’ {aka ‘int’}
  144 |         if(m_axisId.isXAxis()) {
      |                     ^~~~~~~
/var/cache/makepkg/build/scopy-git/src/scopy/gui/src/axishandle.cpp:155:29: error: request for member ‘isXAxis’ in ‘((scopy::AxisHandle*)this)->scopy::AxisHandle::m_axisId’, which is of non-class type ‘QwtAxisId’ {aka ‘int’}
  155 |                 if(m_axisId.isXAxis()) {
      |                             ^~~~~~~
/var/cache/makepkg/build/scopy-git/src/scopy/gui/src/axishandle.cpp: In member function ‘QRect scopy::AxisHandle::getSmallRect()’:
/var/cache/makepkg/build/scopy-git/src/scopy/gui/src/axishandle.cpp:171:21: error: request for member ‘isXAxis’ in ‘((scopy::AxisHandle*)this)->scopy::AxisHandle::m_axisId’, which is of non-class type ‘QwtAxisId’ {aka ‘int’}
  171 |         if(m_axisId.isXAxis()) {
      |                     ^~~~~~~
/var/cache/makepkg/build/scopy-git/src/scopy/gui/src/axishandle.cpp:180:29: error: request for member ‘isXAxis’ in ‘((scopy::AxisHandle*)this)->scopy::AxisHandle::m_axisId’, which is of non-class type ‘QwtAxisId’ {aka ‘int’}
  180 |                 if(m_axisId.isXAxis()) {
      |                             ^~~~~~~
/var/cache/makepkg/build/scopy-git/src/scopy/gui/src/axishandle.cpp: In member function ‘QLine scopy::AxisHandle::getLine()’:
/var/cache/makepkg/build/scopy-git/src/scopy/gui/src/axishandle.cpp:198:21: error: request for member ‘isXAxis’ in ‘((scopy::AxisHandle*)this)->scopy::AxisHandle::m_axisId’, which is of non-class type ‘QwtAxisId’ {aka ‘int’}
  198 |         if(m_axisId.isXAxis()) {
      |                     ^~~~~~~
/var/cache/makepkg/build/scopy-git/src/scopy/gui/src/axishandle.cpp: In member function ‘QRect scopy::AxisHandle::getRectFromLine(QLine)’:
/var/cache/makepkg/build/scopy-git/src/scopy/gui/src/axishandle.cpp:209:21: error: request for member ‘isXAxis’ in ‘((scopy::AxisHandle*)this)->scopy::AxisHandle::m_axisId’, which is of non-class type ‘QwtAxisId’ {aka ‘int’}
  209 |         if(m_axisId.isXAxis()) {
      |                     ^~~~~~~
/var/cache/makepkg/build/scopy-git/src/scopy/gui/src/axishandle.cpp: In member function ‘void scopy::AxisHandle::updateHandleOrientation()’:
/var/cache/makepkg/build/scopy-git/src/scopy/gui/src/axishandle.cpp:226:21: error: request for member ‘isXAxis’ in ‘((scopy::AxisHandle*)this)->scopy::AxisHandle::m_axisId’, which is of non-class type ‘QwtAxisId’ {aka ‘int’}
  226 |         if(m_axisId.isXAxis()) {
      |                     ^~~~~~~
/var/cache/makepkg/build/scopy-git/src/scopy/gui/src/axishandle.cpp: In member function ‘void scopy::AxisHandle::onMouseMove(QPointF)’:
/var/cache/makepkg/build/scopy-git/src/scopy/gui/src/axishandle.cpp:271:21: error: request for member ‘isXAxis’ in ‘((scopy::AxisHandle*)this)->scopy::AxisHandle::m_axisId’, which is of non-class type ‘QwtAxisId’ {aka ‘int’}
  271 |         if(m_axisId.isXAxis()) {
      |                     ^~~~~~~
/var/cache/makepkg/build/scopy-git/src/scopy/gui/src/axishandle.cpp: In member function ‘void scopy::AxisHandle::onResize()’:
/var/cache/makepkg/build/scopy-git/src/scopy/gui/src/axishandle.cpp:326:21: error: request for member ‘isXAxis’ in ‘((scopy::AxisHandle*)this)->scopy::AxisHandle::m_axisId’, which is of non-class type ‘QwtAxisId’ {aka ‘int’}
  326 |         if(m_axisId.isXAxis()) {
      |                     ^~~~~~~
/var/cache/makepkg/build/scopy-git/src/scopy/gui/src/axishandle.cpp: In member function ‘void scopy::AxisHandle::onDoubleClick(QPointF)’:
/var/cache/makepkg/build/scopy-git/src/scopy/gui/src/axishandle.cpp:336:29: error: request for member ‘isXAxis’ in ‘((scopy::AxisHandle*)this)->scopy::AxisHandle::m_axisId’, which is of non-class type ‘QwtAxisId’ {aka ‘int’}
  336 |                 if(m_axisId.isXAxis()) {
      |                             ^~~~~~~
make[2]: *** [gui/CMakeFiles/scopy-gui.dir/build.make:311: gui/CMakeFiles/scopy-gui.dir/src/axishandle.cpp.o] Error 1
make[2]: Leaving directory '/var/cache/makepkg/build/scopy-git/src/build'
make[1]: *** [CMakeFiles/Makefile2:1732: gui/CMakeFiles/scopy-gui.dir/all] Error 2
make[1]: Leaving directory '/var/cache/makepkg/build/scopy-git/src/build'
make: *** [Makefile:169: all] Error 2
```

GCC version is 15.1.1.

Regards!

AlexandraTrifan, 2025-05-19:
Hi,

What version of libqwt do you have installed? Did you install it using a package manager?
Scopy uses a separate libqwt branch: qwt-multiaxes-updated . ( https://github.com/analogdevicesinc/scopy/blob/main/ci/ubuntu/ubuntu_build_process.sh#L80 )
Not using that branch, but the main one instead, might be the cause of the issues you are seeing.
I recommend using the build script provided above. It might require some adjustments to account for packages installed via the package manager if you are using a different Linux distribution. However, it provides clear steps for building all the necessary dependencies.

-Alexandra

dreirund, 2025-05-19:
> *What version of libqwt do you have installed?*

* Qt6 qwt: 6.3.0
* Qt5 qwt would be 5.2.3

> *Did you install it using a package manager?*

Of course.

> *Scopy uses a separate libqwt branch:*

Here I give and gave up. I am not going to use and strange script to compile software.

Closing as not applicable because I do not follow specialised build requirements.

Regards!